256foundation/emberone00-pcb #18: Double check BM1362 footprint spacing

skot, 2025-02-25:
It seems like the BM1362 pins extend a little too close to the center power pads. Double check the spacing with the S19j Pro hashboard footprint

skot, 2025-03-11:
In 67fa04b I changed the spacing between the pads and the pins and moved the VDD pad up a little bit.

I think there still might be an issue with the pin pitch.

econoalchemist, 2025-03-12:
I see three potential variables that could complicate efforts to ensure the spacing is correct:

1) The pin spacing from chip to chip might vary. Could be worth measuring two or more chips to test this theory. 

2) The PCB manufacturer may have a loose tolerance that misplaced some of the pin pads. Could be worth measuring the PCB itself to ensure it matches the requested design measurements. 

3) The required precision might be beyond the resolution capabilities of the imaging equipment. Could be helpful to sharpen the image in GIMP or Photoshop or an online tool prior to feeding the image to the measuring software to mitigate aliasing.

skot, 2025-04-09:
fixed in v3

closed 2025-04-09 by skot