256foundation/emberone00-pcb #37: Add reverse polarity protection?

skot, 2025-04-24:
Right now the emberOne is unprotected from reverse polarity power connections. I kinda think that's fine, but it would be nice to not destory your computer too.. maybe we need reverse polarity protestion on the USB port? for the upstream host?

zbomz, 2025-05-02:
Based on my experience with Loki plebs tinkering with Loki rigs and seeing about half of them blow up at least one hash board while tinkering, I'm a huge proponent of adding reverse polarity protection.

As discussed in the "Ember One Schematic Review Video," reverse polarity protection could be added to either the USB port to at least protect the host PC or to the hash board's main power terminals. Adding reverse polarity protection to the USB port presents a challenge given that both power (VBUS) and data are connected. VBUS can be protected fairly easily with a simple shottky diode given the sub 500mA current draw, but protecting the high speed data lines is harder. It would likely require expensive isolation components/circuitry, and after all that, the hash board would still be vulnerable.    

On the other hand, adding reverse protection polarity to the hash board's main power terminals should protect both the hash board and the host PC, and robust reverse polarity protection can be achieved relatively simply and cheap. 

The design shown below implements an "ideal diode" circuit using an N-type MOSFET on the high side rail and a dedicated charge pump driver. This design offers reverse polarity protection while limiting the forward voltage drop to no more than 35mV at 10 amps. Components were selected specifically to support the max 24V input voltage. The component cost at 1K quantities is about $1.10. 

[image]

skot, 2025-06-09:
U1: AP74700QW6-7 [DK: [31-AP74700QW6-7CT-ND](https://www.digikey.com/en/products/detail/diodes-incorporated/ap74700qw6-7/16680522)]
Q1: NVMFS5C645NL [DK: [488-NVMFS5C645NLAFT1GCT-ND](https://www.digikey.com/en/products/detail/onsemi/NVMFS5C645NLAFT1G/7220982)]
C_CP: CC0805KRX7R9BB104
CIN: GRM155R71H223KA12J
COUT: GRM21BR61H745ME51L
TVS: SMBJ30CA [DK: [SMBJ30CALFCT-ND](https://www.digikey.com/en/products/detail/littelfuse-inc/SMBJ30CA/286005)]